hhd-dev/hhd #120: Why is the default mapping required for GPD back buttons?

shade-belisar, 2024-12-08:
I'm running Bazzite on a GPD Win MAX 2 (2024). To open the HHD UI  (and also the general Steam UI) I need to keep the configuration for the back buttons at the default. However, I would like to be able to actually use these buttons in-game. But when I remap them (e.g. left to X, right to A) long-pressing does not open HHD anymore.

My question is: Why is it necessary to use the default mapping here? And is there some recommended way to be able to both open the UI and use the buttons in-game?

antheas, 2024-12-08:
Yeah if you use hhd and switch to steam controller emulation you can use steam input to remap the buttons and only select one of them for opening menus

It's not intuitive currently gpd needs a rework

shade-belisar, 2024-12-10:
Thanks, that's already an improvement. However, I would like to be able to use both back buttons in-game. Ideally I'm looking to map all three panels to the "Menu" button: The steam quick settings to one click, hhd to long press and the left (home?) menu to double press. Is that possible? I'm not quite sure how the right back button is currently mapped to start hhd because I can't seem to find it in the steam controller configuration, so I don't know if it can be changed.

antheas, 2024-12-10:
It's one of the controller options it's easy to find

Yes the plan is to do multiplexing for a lot of options