hhd-dev/hhd #121: HHD on AUR doesn't show up overay on Steam Game mode.

noizerpark, 2024-12-17:
I'm using CachyOS on ASUS ROG ALLY now. I deleted InputPlumber. HHD can adjust TDP and Controller and LED.
but When I install by "yay -S aur/hhd aur/adjustor aur/hhd-ui" (because old hhd package is on CachyOS's Repo), It installed and work on desktop mode. but in Steam Game Mode(gamescope session) hhd-ui doesn't show up. but setting is still fine.

So, I installed hhd using Curl. It works fine with gamescope session. hhd-ui works fine.
but hhd can't update themself with Error 26 Text file busy: '$HOME/.local/bin/hhd-ui'

What is the easy problem to solve?
(actually I know the answer. back to the bazzite lol)

antheas, 2024-12-17:
Mmmm it should always show up in gaming mode. It would be good to check in journalctl why the AUR package does not work

As for the local updater, while not a priority I will try to look into it

noizerpark, 2024-12-18:
I don't know what is difference, but after reinstall, suddenly Aur version works on Gaming mode. problem solved.

closed 2024-12-18 by noizerpark