hhd-dev/hhd #128: HHD does not start on bazzite-deck-nvidia

SimpleHeuristics, 2025-01-09:
Gamescope session seems to be working decently now in the latest testing branches of deck-nvidia.

However HHD does not appear to start. When in desktop mode and running HHD the UI appears but asks if the application is running.

Running hhd via the terminal then allows the UI in desktop mode to interact with it and change settings.

Going back into game mode HHD also doesn't run. I can SSH into the PC and start HHD via the shell but then it gives errors about how hhd ui fails to start.

I know HHD is not designed for desktop use but it was still nice to be able to update from there and also change performance profiles.

antheas, 2025-01-09:
`sudo systemctl start hhd@$(whoami)` Hopefully no overlay issues. I will note that external controller support is experimental.

SimpleHeuristics, 2025-01-10:
With that command it starts and allows the overlay back in gaming mode. The overlay has no issues so long as the Gamescope session also has no artifacts.

That command doesn't make it work at each boot however. Either have to SSH in to run it or go to desktop mode first. Trying to enable it as a service as I have for other systems services doesn't work.

I think in the future being able to have the HHD menu even for desktop builds would be amazing for rolling back without having to enter KDE and the easy bug reports.

antheas, 2025-01-10:
Sorry, `sudo systemctl enable --now `

closed 2025-01-16 by SimpleHeuristics

SimpleHeuristics, 2025-07-29:
With the new external controller support in HHD will it be enabled by default for deck builds on a desktop or will we still have to enable the service like this?

Also for non-handhelds are there any plan to integrate functionality with LACT? That way dGPUs can be overclocked / undervolted / power limited while in Gamescope session more easily.