hhd-dev/hhd #129: Feature request: Add checkboxes to invert Pitch and Yaw

TurboK234, 2025-01-11:
I have GPD Win 4 (the original 6800U version), and Bazzite (latest) installed.

HHD is great and an absolute requirement for a handheld! Glad to see GPD getting attention lately, e.g. with the v. 3.9.0 updates!

The one thing that I'm struggling with is gyro. Especially when used outside of Steam input. Here's the issue:

Using Dualshock setting with "Default" motion axis setting the controller works normally within an application that uses sdl motion (steam input disabled). The problem is that the game I'm playing is designed to be played with inverted pitch (x axis), so that you can use it for aiming as if looking through the screen. And Dualsense is working just the opposite. I guess it is the "proper" way for gyro to work, but the as a result rotating the screen/controller up makes the player look down, which is counterintuitive and not the way the game is designed.

I am able to set the axis to work properly when I use the "Override" setting for Motion Axis (in this case: Axis X -> X, Invert X Checked; Axis Y -> Y, Invert Y Checked; Axis Z -> Z, Invert Z not checked. The info shows Manufacturer "GPD" and Product "G1618-04").

When using the Default setting, the gyro position stays calibrated and seems to autocalibrate EVEN if I open HHD overlay (you can see drifting happening as soon is the HHD overlay is opened, but it gets stabilized in a few seconds after focus is returned). So the default axis setting seem to work very well (just the pitch is inverted).

The problem is this: when using Override settings, the gyro is instantly decalibrated when I open HHD overlay and the application shows strong and permanent diagonal drift. This drift does not stop after I return to the application and the calibration needs to be reset from application settings for the motion controls to be usable.

Of note: when I toggle the "Invert Roll Axis" with Default Motion Axis, the Roll Axis is perfectly inverted and the calibration is maintained. Roll axis is not the one I need to adjust, but the inversion seems to work perfectly when using this method and it does not mess the calibration.

So as a feature request, either:

Add checkboxes for each axis, i.e. "Invert Pitch Axis" and "Invert Yaw Axis" to accompany "Invert Roll Axis", since the method HDD uses for this inversion seems least intrusive for the axis calibration. I would prefer this solution, since (emulated) games tend to be designed with different axis orientations in mind, and this would be the fastest way to adjust the gyro to match the game.

or

Fix the difference how the Default and Override take control of Dualshock gyro emulation. Fixing this could also be considered fixing a bug (and this issue as a bug report), but the first suggestion would skip the need to change other (possibly intended) HHD behaviour.

antheas, 2025-01-11:
Currently, I think we invert the Roll axis only on the gyro side and leave  the accelerometer the same. Is that the problem?

antheas, 2025-01-11:
As for the calibration being messed up when you open the overlay, yes we need to do some work there. HHD will add noise to the accelerometer values to mess with applications that try to recalibrate, as we stop passing through the gyro values. Maybe we should pass them through instead.

TurboK234, 2025-01-11:
Regarding Roll Axis and gyro vs. accel, what I found when browsing the code, the flip_z seems to only change the gyro values, copy-pasted from the code:

MOTION_AXIS_MAP: dict[Axis, AX] = {
    "accel_x": AX(B("ABS_X"), 8192 / 9.8, bounds=(-32768, 32768)),
    "accel_y": AX(B("ABS_Y"), 8192 / 9.8, bounds=(-32768, 32768)),
    "accel_z": AX(B("ABS_Z"), 8192 / 9.8, bounds=(-32768, 32768)),
    "gyro_x": AX(B("ABS_RX"), 1024 * 180 / 3.14, bounds=(-2097152, 2097152)),
    "gyro_y": AX(B("ABS_RY"), 1024 * 180 / 3.14, bounds=(-2097152, 2097152)),
    "gyro_z": AX(B("ABS_RZ"), 1024 * 180 / 3.14, bounds=(-2097152, 2097152)),
}

MOTION_AXIS_MAP_FLIP_Z: dict[Axis, AX] = {
    "accel_x": AX(B("ABS_X"), 8192 / 9.8, bounds=(-32768, 32768)),
    "accel_y": AX(B("ABS_Y"), 8192 / 9.8, bounds=(-32768, 32768)),
    "accel_z": AX(B("ABS_Z"), 8192 / 9.8, bounds=(-32768, 32768)),
    "gyro_x": AX(B("ABS_RX"), 1024 * 180 / 3.14, bounds=(-2097152, 2097152)),
    "gyro_y": AX(B("ABS_RY"), 1024 * 180 / 3.14, bounds=(-2097152, 2097152)),
    "gyro_z": AX(B("ABS_RZ"), -1024 * 180 / 3.14, bounds=(-2097152, 2097152)),

But as of fixing my issue, I don't know if considering/ignoring the accelerometer is relevant, and maybe just shows how I do not really understand the bigger picture of motion controls. Maybe this is irrelevant for the issue itself, but just an observation that whatever happens here is not intrusive for the application.

antheas, 2025-01-11:
Invert roll Axis is there because when you use a controller vs a handheld to aim, you turn the roll axis the other way. Ideally, it would be fixed in Steam Input or in the emulators by specifying that you are using a handheld, but we are not there yet.

The roll axis is the one where if you have a controller facing up (e.g., the touchpad aims to the ceiling) you would turn it like a steering wheel. If you have a handheld and the screen is pointing to your face, its the axis that is the same as turning your head.

If you have a mind experiment, you will see that how you would use those is flipped.

The other options are always correct, as you use them the same way.

antheas, 2025-01-11:
Try flipping the accelerometer too to see if that helps.

TurboK234, 2025-01-11:
Yes, I do see now how the "handheld/steering wheel roll" becomes the yaw in gamepad orientation, and "gamepad roll" becomes yaw when lifted 90 degrees to handheld orientation... and that the Roll Axis Invert is meant to compensate for that, which I too appreciate.

But for me, pitch is still the real issue, which requires me to use Override in Axis.

I kind of got lost on your last comment, what do you mean by flipping the accelerometer?

TurboK234, 2025-01-11:
After doing some more testing I see that the main problem is the the calibration that gets off every time I visit HHD:s overlay (especially when using Override), so it's difficult to do the adjustments for current setup (handheld mode/external gyro-enabled controller etc). Especially since the calibration is very well hidden in certain emulators.
So if there is a way to let the gyro values pass through with a patch... maybe it would help.

antheas, 2025-01-11:
I will look into skipping that when the overlay is open, my focus when building that was to stop steam from recalibrating

If your pitch is wrong, post your values in the override so i can update your matrix. Just make sure to follow the instructions in the main readme to see that they are correct. Ie with invert roll axis and 3d to 2d being Local space everything works as expected.

The GPD Win 4 has a different axis between 7800u and 8840u, maybe 6800u is different as well

TurboK234, 2025-01-12:
Thanks for taking this a look at this!
I followed the instructions in Readme to match the axis and the Steam gyro calibration glyphs and played some Half-Life 2 using the "Camera" setting (and finally switched the turning axis). Using these settings the gyro movement seems correct. It would sure be easiest, if HHD would use these settings as default for the device :D 

[image]

antheas, 2025-01-12:
Try the beta button in bug report https://github.com/hhd-dev/hhd/commit/40af56e8fe4624011caace4bda80a8a6b21551b5

TurboK234, 2025-01-13:
I tested the beta version. Observations:

-Steam Gyro Calibration screen (and the glyph movements) are consistent with the actual movement of the device, when using "Default" Motion Axis in HHD
-Half Life 2 works now perfectly with HHD "Default" Motion Axis (left the Gyro Turning Axis to Roll, as instructed in the Readme)

So the gyro axis are now in correct orientation for GPD Win 4 6800U by default.

Then the weird ones:
-In an emulator, the motion acts strangely, even more so than before. In aiming situation, the aim may be pointing horizontally in neutral position _at first_, but then it heavily drifts to point either north or south. This is not caused by HHD Overlay, since it happens when only using the emulator.
-The actual movement seems to work as intended (pitch orientation etc) for a short period, but motion is unusable. Again, I can bypass the Default settings with Override, and this heavy drifting ends.
-HDD overlay still messes the calibration in emulation, but not in Steam game / Half-life 2, I guess Steam takes care of the calibration... or maybe steam input is not dependent on the accelerometer so much. And I know, this beta fix was not intended to fix the Overlay/calibration issue. 

The feeling I get is that the motion (when using both gyro and accelerometer) is getting conflicting input values that end up polarizing the orientation either up of down.

antheas, 2025-01-13:
So we have a small problem. I think the calibration was correct for the 6800U, so we might find a different way to differentiate between the units. I need to look into it.

antheas, 2025-01-22:
@parth-menon find a way to differentiate between your devices. I pushed this and shouldnt have if you are on the latest version

parth-menon, 2025-01-22:
> [@parth-menon](https://github.com/parth-menon) find a way to differentiate between your devices. I pushed this and shouldnt have if you are on the latest version

This is unfortunate. Let me dig into system info and revert back.

TurboK234, 2025-01-24:
Regarding the drifting and calibration problem when bringing up HHD overlay, I can confirm it is now fixed with the latest commits (supposedly [](https://github.com/hhd-dev/hhd/commit/2b94211a5bbcc822ca67c25f9139d3becb71ec88) )

No apparent drawbacks, now the emulated games stay calibrated and get the gyro data even when I bring up the HHD overlay. Steam games do not seem to mind, either. Because of this, it is easier to switch between Default and Override (without calibrating in between). 

Default still has wrong axis configuration with my device, I see that GPD device recognition is being worked on.

One thing to mention is that Default seems to keep the axis orientation very well (i.e. the orientation stays horizontal, vertical and z in 90 degree angles with the device, albeit unintuitively oriented atm), whereas when using Override, the axis can be set to match the movement, but axis seems to have a tendency to become a bit diagonal after a while. Meaning that when I pitch up, the view moves up _and left_ diagonally and vice versa, pitching down looks down _and right_ diagonally.

Maybe this is intended, maybe Override also overrides something the sensor/device does to keep itself oriented. Or maybe GPD's gyro is not great at keeping the gyro in line. Anyway, this shall be a non-issue when the GPD Win 4 6800U gets the correct axis configuration with Default settings, no need to use the Override after that.

antheas, 2025-01-24:
I actually fixed the override for you, but I reverted it because I have at least 3 6800Us with a different config

Soi you guys need to:
```
for f in /sys/class/dmi/id/*; do echo "$f:"; sudo cat $f; done
```

And compare and contrast

parth-menon, 2025-01-24:
> I actually fixed the override for you, but I reverted it because I have at least 3 6800Us with a different config
> 
> Soi you guys need to:
> 
> ```
> for f in /sys/class/dmi/id/*; do echo "$f:"; sudo cat $f; done
> ```
> 
> And compare and contrast

Here is mine. GPD Win4 7840U
                                                                                
steam@bazzite:~$ for f in /sys/class/dmi/id/*; do echo "$f:"; sudo cat $f; done
/sys/class/dmi/id/bios_date:
[sudo] password for steam:      
07/19/2024
/sys/class/dmi/id/bios_release:
0.60
/sys/class/dmi/id/bios_vendor:
American Megatrends International, LLC.
/sys/class/dmi/id/bios_version:
0.60
/sys/class/dmi/id/board_asset_tag:
Default string
/sys/class/dmi/id/board_name:
G1618-04
/sys/class/dmi/id/board_serial:
Default string
/sys/class/dmi/id/board_vendor:
GPD
/sys/class/dmi/id/board_version:
Ver. 1.0
/sys/class/dmi/id/chassis_asset_tag:
Default string
/sys/class/dmi/id/chassis_serial:
Default string
/sys/class/dmi/id/chassis_type:
10
/sys/class/dmi/id/chassis_vendor:
SU
/sys/class/dmi/id/chassis_version:
Default string
/sys/class/dmi/id/ec_firmware_release:
0.21
/sys/class/dmi/id/modalias:
dmi:bvnAmericanMegatrendsInternational,LLC.:bvr0.60:bd07/19/2024:br0.60:efr0.21:svnGPD:pnG1618-04:pvrVer.1.0:rvnGPD:rnG1618-04:rvrVer.1.0:cvnSU:ct10:cvrDefaultstring:skuDefaultSKU:
/sys/class/dmi/id/power:
cat: /sys/class/dmi/id/power: Is a directory
/sys/class/dmi/id/product_family:
Game machining Series 27
/sys/class/dmi/id/product_name:
G1618-04
/sys/class/dmi/id/product_serial:
Default string
/sys/class/dmi/id/product_sku:
Default SKU
/sys/class/dmi/id/product_uuid:
27bf5f80-2d9e-11ee-a627-d706275b5b0b
/sys/class/dmi/id/product_version:
Ver.1.0
/sys/class/dmi/id/subsystem:
cat: /sys/class/dmi/id/subsystem: Is a directory
/sys/class/dmi/id/sys_vendor:
GPD
/sys/class/dmi/id/uevent:
MODALIAS=dmi:bvnAmericanMegatrendsInternational,LLC.:bvr0.60:bd07/19/2024:br0.60:efr0.21:svnGPD:pnG1618-04:pvrVer.1.0:rvnGPD:rnG1618-04:rvrVer.1.0:cvnSU:ct10:cvrDefaultstring:skuDefaultSKU:

antheas, 2025-01-24:
God damn it, your's are the 8840U mappings we thought

parth-menon, 2025-01-24:
> God damn it, your's are the 8840U mappings we thought

So the 7840U and 8840U have the same mappings? 
Is it just the 6800U that's different?

antheas, 2025-01-24:
No, Turbo has a 6800U with different values than yours

Then there are 7840Us and 8840Us with different mappings. You have what we thought was 8840U mappings

TurboK234, 2025-01-25:
Here's the list of my /sys/class/dmi/id/* 
For reference GPD Win 4 6800U, 16 Gb RAM, Originally 1 Tb HDD. 

And the proper way to set the axis in Override (at the moment) is
Axis X -> X Invert X Checked
Axis Y -> Z Invert Y Checked
Axis Z -> Y Invert Z Not Checked

`/sys/class/dmi/id/bios_date:
06/06/2023
/sys/class/dmi/id/bios_release:
5.24
/sys/class/dmi/id/bios_vendor:
American Megatrends International, LLC.
/sys/class/dmi/id/bios_version:
3.06
/sys/class/dmi/id/board_asset_tag:
Default string
/sys/class/dmi/id/board_name:
G1618-04
/sys/class/dmi/id/board_serial:
Default string
/sys/class/dmi/id/board_vendor:
GPD
/sys/class/dmi/id/board_version:
Default string
/sys/class/dmi/id/chassis_asset_tag:
Default string
/sys/class/dmi/id/chassis_serial:
Default string
/sys/class/dmi/id/chassis_type:
30
/sys/class/dmi/id/chassis_vendor:
Default string
/sys/class/dmi/id/chassis_version:
Default string
/sys/class/dmi/id/ec_firmware_release:
8.0
/sys/class/dmi/id/modalias:
dmi:bvnAmericanMegatrendsInternational,LLC.:bvr3.06:bd06/06/2023:br5.24:efr8.0:svnGPD:pnG1618-04:pvrDefaultstring:rvnGPD:rnG1618-04:rvrDefaultstring:cvnDefaultstring:ct30:cvrDefaultstring:skuDefaultstring:
/sys/class/dmi/id/power:
cat: /sys/class/dmi/id/power: On hakemisto
/sys/class/dmi/id/product_family:
Default string
/sys/class/dmi/id/product_name:
G1618-04
/sys/class/dmi/id/product_serial:
Default string
/sys/class/dmi/id/product_sku:
Default string
/sys/class/dmi/id/product_uuid:
03000200-0400-0500-0006-000700080009
/sys/class/dmi/id/product_version:
Default string
/sys/class/dmi/id/subsystem:
cat: /sys/class/dmi/id/subsystem: On hakemisto
/sys/class/dmi/id/sys_vendor:
GPD
/sys/class/dmi/id/uevent:
MODALIAS=dmi:bvnAmericanMegatrendsInternational,LLC.:bvr3.06:bd06/06/2023:br5.24:efr8.0:svnGPD:pnG1618-04:pvrDefaultstring:rvnGPD:rnG1618-04:rvrDefaultstring:cvnDefaultstring:ct30:cvrDefaultstring:skuDefaultstring:`