hhd-dev/hhd #130: Using extra buttons for a script

maciekish, 2025-01-14:
Hi, i have an Ayaneo Air and im able to bind the LC and RC buttons to show steam keyboard and open hdd, and that works. However sometime the screen is black after rebooting or waking from sleep, and only systemctl restart sddm will fix it. I want to bind a long press of RC to this action to avoid force-restarting the whole device when im not able to ssh in. But im unable to figure out how to capture the button press, how does HHD do it?

antheas, 2025-01-14:
Try disabling `Power off display before sleep` in hhd

There is a feature like that for other devices where holding select will reboot. Maybe I will add it here

Also, when it happens, please send a bug report as we should fix this.

antheas, 2025-01-14:
Does steam make sounds? Please do a bug report

maciekish, 2025-01-14:
> Does steam make sounds? Please do a bug report

Yes, it does. Only the screen is dead and comes to life with `systemctl restart sddm`. Bug report in Bazzite repo i guess?

antheas, 2025-01-14:
Generate a bug report using handheld daemon's bug report tool. Does not matter where you paste it

maciekish, 2025-01-14:
I skimmed through it quickly, my manual restart of sddm is at line 4818, and only after that is there `Starting Wayland user session`. Is it not attempting to start at all before that?

https://paste.centos.org/view/ac2535ad

antheas, 2025-01-14:
Your remote desktop software might be causing issues

maciekish, 2025-01-14:
> Your remote desktop software might be causing issues

I had the issue on a fresh install, i installed tigervnc and enabled RDP in KDE to troubleshoot easier. Anyway i have removed tigervnc now and RDP doesnt start in game mode so that shouldn't be the problem.

antheas, 2025-01-14:
Post a new bug report with no crashes then

maciekish, 2025-01-14:
By no cashes you mean a successful boot without tigervnc or a black screen boot without tigervnc?

maciekish, 2025-01-14:
Successful boot after removing everything except htop, glances & syncthing: https://paste.centos.org/view/7a184438
Black screen boot with the same packages as the successful boot: https://paste.centos.org/view/37180826

All installed packages are pure terminal tools that have nothing to do with x, wayland etc. The black screen boot has a manual systemctl restart sddm issued via ssh to get the screen back.

maciekish, 2025-01-14:
This is a black screen boot without manually restarting sddm, i just rebooted it via ssh once it had booted with a black screen and used last boot (-1) in the bug report tool https://paste.centos.org/view/8fb6d438

maciekish, 2025-01-14:
I guess this is significant

```
Jan 14 22:15:23 xdg-desktop-portal-kde[2673]: qt.qpa.xcb: could not connect to display
Jan 14 22:15:23 audit[2673]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 subj=unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023 pid=2673 comm="xdg-desktop-por" exe="/usr/libexec/xdg-desktop-portal-kde" sig=6 res=1
Jan 14 22:15:23 hhd[1203]: PBTN   INFO     Starting multi-device powerbutton handler for device 'AIR'.
Jan 14 22:15:23 xdg-desktop-portal-kde[2673]: qt.qpa.plugin: From 6.5.0, xcb-cursor0 or libxcb-cursor0 is needed to load the Qt xcb platform plugin.
Jan 14 22:15:23 xdg-desktop-portal-kde[2673]: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Jan 14 22:15:23 xdg-desktop-portal-kde[2673]: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
                                             
                                              Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vkkhrdisplay, vnc, wayland-egl, wayland, xcb.
```

antheas, 2025-01-14:
```
Process 3097 (krdpserver) of user 1000 dumped core.
Process 3294 (xdg-desktop-por) of user 1000 dumped core.
```
I dont like this

I would say ignore the other portal for now. The QT error is also a bit benign. The problem is that these things crash and maybe krdpserver is involved.

maciekish, 2025-01-14:
Okay, i see now that krdp is mentioned even when launching directly into gamescope without desktop mode. I will try to switch to desktop and disable krdp. Thanks.

antheas, 2025-01-14:
Yeah looks normal though. But you have a Renoir unit, thats quite old, let me tag @KyleGospo

maciekish, 2025-01-14:
It's the first Ayaneo Air before they released the refresh. It's about 1.5 years old (in my hand).

maciekish, 2025-01-15:
Hi again, is it possible to add an option to hdd to not grab the keyboard combos exclusively? That way i could just daemonize a python script to look for the LC/RC keyboard combos and run my command alongside hhd.

antheas, 2025-01-15:
Try this https://github.com/hhd-dev/hhd/commit/6e4ebd02f88d34316750267c23ccb9830e4177f9 by pressing the beta button in updates. I ported it over to ayaneo

Also try testing, it has a fix for weird screen sleep bugs

No, its not possible to disable. Users find these settings and break their stuff unfortunately. Shortcuts for front buttons do weird things otherwise.