hhd-dev/hhd #131: Power profiles are not consistent across HHD and KDE desktop on bazzite-deck-nvidia.

SimpleHeuristics, 2025-01-16:
HHD is set to balanced power profile either via the desktop app or under game mode. When switching to the KDE desktop the system tray shows that the power profile is set to power save. This is the same regardless of the power profile set on HHD.

The tuned service and tuned.ppd services are running based on systemctl.

I don't notice any performance regressions but the behaviour is different from my installation on a handheld where these parameters sync up between HHD and KDE.

antheas, 2025-01-17:
Let me look into it soon. I thought I fixed it but I also rarely test that part since most of the testing is done on handhelds.

SimpleHeuristics, 2025-01-24:
Any logs you need to help with it?

antheas, 2025-01-24:
@KyleGospo you want to have a look into this? Its a tuned issue