hhd-dev/hhd #133: [OneXPlayer2Pro] Enabling Motion breaks controller emulation

parth-menon, 2025-01-21:
Hi team,
I installed Bazzite OS on my OneXPlayer2Pro today(21 Jan 2025) which has HHD preinstalled.
I set Controller Emulation to DualSense and enabled Motion controls and immediately, the controller stops responding and later completely disconnects from the UI.
It only occurs when I enable Motion and happens for both DualSense and Edge input. The controller reconnects and starts working again if I disable Motion from swipe access UI.
Here is a video of the issue: https://drive.google.com/file/d/1AaG4n8k5QPzKCgVWn6x4zu2W_llIK5Jo/view?ts=679028df&pli=1

Please let me know if I can provide any support to get this fixed.

antheas, 2025-01-22:
Post a log from hhd bug report menu

antheas, 2025-01-22:
Try to update. Something upsets the gyro of your device

antheas, 2025-01-22:
Maybe it's the frequency

parth-menon, 2025-01-22:
> Try to update. Something upsets the gyro of your device

Update hhd? Or bazzite?

antheas, 2025-01-22:
Bazzite. Also, try other frequencies for the gyro

parth-menon, 2025-01-22:
> Bazzite. Also, try other frequencies for the gyro

Updating Bazzite worked! For future reference, the freq that worked was 400Hz. The default was 200Hz.

Thank you.

antheas, 2025-02-17:
Should be fixed now

closed 2025-02-17 by antheas