hhd-dev/hhd #136: [GPD Win Max 2 2023] DSU Client on port 26760 for gyro usage in emulator

KIDult2226, 2025-01-24:
System: Cachy OS rolling
Hardware: GPD Win Max 2 2023 64GB

I recently noticed that HHD seems capable of detecting motion hardware on GPD devices. 
I was wondering if it would be possible to consider adding a persistent DSU Client that could provide motion data to relevant emulators?

Thank you so much for your hard work on this project!

antheas, 2025-01-24:
We thought about it but pretty much all emulators support the stock hori and dualsense controllers so it would be more complicated

Is there a specific emulator you have in mind?

KIDult2226, 2025-01-28:
> We thought about it but pretty much all emulators support the stock hori and dualsense controllers so it would be more complicated
> 
> Is there a specific emulator you have in mind?

Thank you so much for replying.

The most frequent emulator in my usage would be CEMU and Citron, both of which supports 26760 port for DSU client. I don't seem to have other solution other than enable a client for them to support motions on my GPD devices.

antheas, 2025-01-28:
Try evdev hook. Follow the instructions for a dualsense or joycon controller

KIDult2226, 2025-01-29:
> Try evdev hook. Follow the instructions for a dualsense or joycon controller

Did so, still not picking up any signal from the client.

[image]

[image]

Is GPD's internal motion sensor supported?

antheas, 2025-01-29:
Ok let's start from the beginning. Install bmi260-dkms to enable the internal sensor. Set emulation to dualsense. Use evdev hook to find dualsense controller. Then use CEMU hook

KIDult2226, 2025-01-29:
> Ok let's start from the beginning. Install bmi260-dkms to enable the internal sensor. Set emulation to dualsense. Use evdev hook to find dualsense controller. Then use CEMU hook

Thank you for the quick reply. However I encountered some fatal constructing issues while building it through AUR and git clone:

❯ tail /var/lib/dkms/bmi260/1.0.0/build/make.log
<inline asm>:3:92: error: expected string
    3 | .section ".export_symbol","a" ; __export_symbol_bmi260_core_probe: ; .asciz "GPL" ; .asciz IIO_BMI260 ; .balign 8 ; .quad bmi260_core_probe ; .previous
      |                                                                                             ^
3 errors generated.
make[3]: *** [/usr/lib/modules/6.13.0-2-cachyos/build/scripts/Makefile.build:194: bmi260_core.o] Error 1
make[2]: *** [/usr/lib/modules/6.13.0-2-cachyos/build/Makefile:1997: .] Error 2
make[1]: *** [/usr/lib/modules/6.13.0-2-cachyos/build/Makefile:251: __sub-make] Error 2
make[1]: Leaving directory "/var/lib/dkms/bmi260/1.0.0/build"
make: *** [Makefile:251: __sub-make] Error 2
make: Leaving directory "/usr/lib/modules/6.13.0-2-cachyos/build"

I am wondering whether it has some issue related to the new kernel or the customization from CachyOS to 6.13 kernel

antheas, 2025-01-29:
We have not moved to 6.13 yet so I cannot comment on that. Would be worth reaching out to them

closed 2025-02-07 by KIDult2226