hhd-dev/hhd #138: GPD Win Mini 2024: Motion Support disables controller

N1v0k, 2025-01-31:
If I activate "Motion Support" when selecting either Dual Sense or Steam Controller emulation, the built-in controllers stop working completely.
It was working fine for some time, but since two months it broke.

Any advice?

Krules02, 2025-02-01:
Same issue on 2023 model/bazzite, except it worked yesterday for me (first time I used it/switched from Xbox mode)

antheas, 2025-02-01:
Can I get some logs from handheld Daemon?

Krules02, 2025-02-01:
https://paste.centos.org/view/5d672fa5

antheas, 2025-02-01:
Press use hhd beta until reboot, which is under the bug report button. Then try again and send another log

Krules02, 2025-02-01:
https://paste.centos.org/view/6189c5a5

Controller worked, but was unable to activate gyro in steam input. Steam kept disabling it.

antheas, 2025-02-01:
I did not fix anything haha

Maybe it was the frequency. Try to make it happen again. Otherwise, version .3 will be on the next bazzite stable

Krules02, 2025-02-01:
Yeah, still not working on other frequencies. I'll just wait for the update if you think that would fix it. Thanks for the help though.

antheas, 2025-02-01:
As you say that, you didn't send a log.

Krules02, 2025-02-01:
I've sent two logs? If you mean a log after changing the frequencies on beta, then here: https://paste.centos.org/view/c17120d3

antheas, 2025-02-01:
This log is clean, so the beta works?

Krules02, 2025-02-01:
No, when I turn gyro on in steam input, it still switched back to "none" and had no effect in game. I tried turning it on more times in steam input and made another log: https://paste.centos.org/view/f1ad0944 

I just restarted (so shouldn't be on beta), and the regular controller buttons work despite having gyro enabled, and gyro works in game. I guess the issue is not consistent? Here is a log where everything is working (not beta): https://paste.centos.org/view/373b5e25

antheas, 2025-02-01:
The hhd decky plugin has not been updated since april, you should consider not using it

I only care if the controller disappears completely. And I need a log for that in hhd beta because I made it show where it crashes.

As for switching to none, thats a Steam quirk and unrelated. If you can see the gyro under advanced calibration it works

N1v0k, 2025-02-01:
Hi there!
I activated the beta version, it works there like it was working before. 

I can also map the gyro, no problem. Tried it in game.

Do you need any more input? When will the beta be released? Do I need to update it separately to bazzite?

Thanks for your work!

antheas, 2025-02-01:
We ran a build today, so you should just be able to update.

antheas, 2025-02-06:
Is this still an issue?

N1v0k, 2025-02-06:
No, it has been resolved for me with the latest update! Thanks.

closed 2025-02-06 by N1v0k

Krules02, 2025-02-06:
It's been working great for me too, thank you.