hhd-dev/hhd #139: Steam Menu not opening most of the Time in Dualsense Mode on the Ally X

NightHammer1000, 2025-02-02:
Noticed that, if you are in Dualsense Mode on the Ally X, the Right Steam Menu won't open most of the Time.
Only Spam Pressing opens it some of the Time.

Looking through the Code, and from what i observed, the Button for it has not Hold Event and the Button Press is sent on Release
Maybe the Button Press Length send after release is not quite long enough for the Dualsense Edge Handler in Steam.

It works perfectly well on the Other Controller Modes.

antheas, 2025-02-02:
I just checked and it works fine on my end

NightHammer1000, 2025-02-02:
Of course, now that I have a camera pointed at it ready to record, it won't happen.

I noticed it the most when the CPU is hammered. For example, while running Star Citizen on it.
Also, it happens a lot if I first open the Right Menu and then try to open the Left.

Will try to get a Video of it.

Happened on Bazzite and plain old Arch so far.

NightHammer1000, 2025-02-02:
Okay... This may sound crazy but I think it has something to do with Device Orientation and Angle?

https://drive.google.com/file/d/1KLYHjb9bP3x5kshUeNXKCInaDPt6IUv-/view?usp=sharing

NightHammer1000, 2025-02-02:
Not Angle directly. Noise. If I have the Device on the Cases Kickstand at an Angle it works fine.

Holding it in Hands it does not.

Something to do with the Gyro??

antheas, 2025-02-02:
Ok I misunderstood. You're talking about a steam client bug

Yes at heavy loads there is a chance the steam client may miss it. I have to tune the delay accordingly

If you set up a scenario that I can repro I can tune for that

NightHammer1000, 2025-02-02:
Ohh nope. Something weird is happening. The Left and Right Buttons are activating Gyro somehow. See the attached Video.

I have gyro nowhere in my Steam Input Configs.

https://drive.google.com/file/d/1KLo8u9tVRWNMG_ME_iQucLp15hd6oydv/view?usp=sharing

This only happens in Dualsense Mode.

NightHammer1000, 2025-02-02:
More Info. Some evtest.

The AllyX controller itself definetly sends gyro when the Left or Right button is pressed in addition to what is send in the Virtual Controller. But only in Dualsense mode.
If I switch it over to Steam Controller Mode the Ally X Controller itself does send nothing when I press the  Left or Right button. The Events only show on the Emulated Controller.

```
EV-Test Dualsense Virtual Device:
Event: time 1738527281.293246, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1738527281.293246, -------------- SYN_REPORT ------------
Event: time 1738527281.443316, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1738527281.443316, -------------- SYN_REPORT ------------
```
```
EV-Test Ally X Controller in Dualsense Mode:
Event: time 1738527352.248241, type 3 (EV_ABS), code 3 (ABS_RX), value 2274
Event: time 1738527352.248241, type 3 (EV_ABS), code 4 (ABS_RY), value 2317
Event: time 1738527352.248241, -------------- SYN_REPORT ------------
Event: time 1738527352.250424, type 3 (EV_ABS), code 3 (ABS_RX), value 0
Event: time 1738527352.250424, type 3 (EV_ABS), code 4 (ABS_RY), value 0
Event: time 1738527352.250424, -------------- SYN_REPORT ------------
Event: time 1738527352.252223, type 3 (EV_ABS), code 3 (ABS_RX), value 2274
Event: time 1738527352.252223, type 3 (EV_ABS), code 4 (ABS_RY), value 2317
Event: time 1738527352.252223, -------------- SYN_REPORT ------------
Event: time 1738527352.254275, type 3 (EV_ABS), code 3 (ABS_RX), value 0
Event: time 1738527352.254275, type 3 (EV_ABS), code 4 (ABS_RY), value 0
Event: time 1738527352.254275, -------------- SYN_REPORT ------------
Event: time 1738527352.256294, type 3 (EV_ABS), code 3 (ABS_RX), value 2274
Event: time 1738527352.256294, type 3 (EV_ABS), code 4 (ABS_RY), value 2317
Event: time 1738527352.256294, -------------- SYN_REPORT ------------
Event: time 1738527352.260330, type 3 (EV_ABS), code 3 (ABS_RX), value 2063
Event: time 1738527352.260330, type 3 (EV_ABS), code 4 (ABS_RY), value 2230
Event: time 1738527352.260330, -------------- SYN_REPORT ------------
Event: time 1738527352.264245, type 3 (EV_ABS), code 3 (ABS_RX), value 1997
Event: time 1738527352.264245, type 3 (EV_ABS), code 4 (ABS_RY), value 2158
Event: time 1738527352.264245, -------------- SYN_REPORT ------------
Event: time 1738527352.270255, type 3 (EV_ABS), code 3 (ABS_RX), value 1828
Event: time 1738527352.270255, type 3 (EV_ABS), code 4 (ABS_RY), value 2110
Event: time 1738527352.270255, -------------- SYN_REPORT ------------
Event: time 1738527352.274237, type 3 (EV_ABS), code 3 (ABS_RX), value 1825
Event: time 1738527352.274237, type 3 (EV_ABS), code 4 (ABS_RY), value 1994
Event: time 1738527352.274237, -------------- SYN_REPORT ------------
Event: time 1738527352.278241, type 3 (EV_ABS), code 3 (ABS_RX), value 0
Event: time 1738527352.278241, type 3 (EV_ABS), code 4 (ABS_RY), value 0
Event: time 1738527352.278241, -------------- SYN_REPORT ------------
Event: time 1738527352.280232, type 3 (EV_ABS), code 3 (ABS_RX), value 1818
Event: time 1738527352.280232, type 3 (EV_ABS), code 4 (ABS_RY), value 1994
Event: time 1738527352.280232, -------------- SYN_REPORT ------------
Event: time 1738527352.284241, type 3 (EV_ABS), code 3 (ABS_RX), value 2250
Event: time 1738527352.284241, -------------- SYN_REPORT ------------
Event: time 1738527352.288246, type 3 (EV_ABS), code 3 (ABS_RX), value 2043
Event: time 1738527352.288246, type 3 (EV_ABS), code 4 (ABS_RY), value 2802
Event: time 1738527352.288246, -------------- SYN_REPORT ------------
Event: time 1738527352.296268, type 3 (EV_ABS), code 3 (ABS_RX), value 2371
Event: time 1738527352.296268, type 3 (EV_ABS), code 4 (ABS_RY), value 2793
Event: time 1738527352.296268, -------------- SYN_REPORT ------------
Event: time 1738527352.298269, type 3 (EV_ABS), code 3 (ABS_RX), value 2144
Event: time 1738527352.298269, type 3 (EV_ABS), code 4 (ABS_RY), value 3527
Event: time 1738527352.298269, -------------- SYN_REPORT ------------
Event: time 1738527352.308312, type 3 (EV_ABS), code 3 (ABS_RX), value 2085
Event: time 1738527352.308312, type 3 (EV_ABS), code 4 (ABS_RY), value 3939
Event: time 1738527352.308312, -------------- SYN_REPORT ------------
Event: time 1738527352.314288, type 3 (EV_ABS), code 3 (ABS_RX), value 2017
Event: time 1738527352.314288, type 3 (EV_ABS), code 4 (ABS_RY), value 4270
Event: time 1738527352.314288, -------------- SYN_REPORT ------------
Event: time 1738527352.318261, type 3 (EV_ABS), code 3 (ABS_RX), value 1955
Event: time 1738527352.318261, -------------- SYN_REPORT ------------
Event: time 1738527352.324291, type 3 (EV_ABS), code 3 (ABS_RX), value 1902
Event: time 1738527352.324291, type 3 (EV_ABS), code 4 (ABS_RY), value 4267
Event: time 1738527352.324291, -------------- SYN_REPORT ------------
Event: time 1738527352.328252, type 3 (EV_ABS), code 3 (ABS_RX), value 1847
[... 83 lines cut ...]
Event: time 1738527352.440269, type 3 (EV_ABS), code 4 (ABS_RY), value -1892
Event: time 1738527352.440269, -------------- SYN_REPORT ------------
Event: time 1738527352.444298, type 3 (EV_ABS), code 3 (ABS_RX), value 2651
Event: time 1738527352.444298, type 3 (EV_ABS), code 4 (ABS_RY), value -1738
Event: time 1738527352.444298, -------------- SYN_REPORT ------------
Event: time 1738527352.448340, type 3 (EV_ABS), code 3 (ABS_RX), value 2637
Event: time 1738527352.448340, type 3 (EV_ABS), code 4 (ABS_RY), value -1604
[... cut ...]
Event: time 1738527352.456296, type 3 (EV_ABS), code 4 (ABS_RY), value 0
Event: time 1738527352.456296, -------------- SYN_REPORT ------------

```
```
EV-Test AllyX Controller in Steam Controller Mode:
-- NOTHING --
```
```
EV-Test Steam Controlle Virtual Device
Event: time 1738527452.274622, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1738527452.274622, -------------- SYN_REPORT ------------
Event: time 1738527452.424583, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1738527452.424583, -------------- SYN_REPORT ------------

```

antheas, 2025-02-02:
Dualsense is uhid. If you do `sudo hhd.contrib hidraw` you will see that pressing the menu buttons does nothing

NightHammer1000, 2025-02-02:
But it does something. Have you seen the Vid?

Just because it does not send anything on the Ally X Controllers itself does not mean it does not send something on the Emulated XBOX 360 Controller.
Because, like we see in the evtest, it does.

From what I understand, the Inputs from the Emulated XBOX 360 Controller should be completely hidden yes?
Well they are not in Dualsense mode. And somehow, in Dualsense mode. When I press the Left or Right Special Keys something is triggerd that translates Gyro to the Right Stick Input on the Emulated XBOX 360 Controller.

The Gyro Input does not come from the Emulated Dualsense itself.

NightHammer1000, 2025-02-02:
Video of the Gamepad Tester:
https://drive.google.com/file/d/1KfoNAX0ZWJUVZj2qE198bR7clPQSCwAh/view?usp=sharing

antheas, 2025-02-02:
> The Gyro Input does not come from the Emulated Dualsense itself. The Input comes from /dev/input/event24: Microsoft X-Box 360 pad 0 And can be seen from Test tools like the Gamepad Test Website. And this also Explains the Steam Issue, as it has to switch between the XBOX Contoller and PS5 Controller Inputs and there the Steam Button press simply gets lost between the Waves of Gyro Input.

What im seeing is that steam is cooked if I am correct. I made sure handheld daemon works correctly, what can i do with that?

NightHammer1000, 2025-02-02:
AH! Yep. You are completly right.

In Steam for Deck, you cannot set a config for the Guide Button, The Option for it is not there. On PC you can.
And guess what. Those Settings are Account Bound, not Device Bound. 

So my Guide Button setting of my Dualsense Edge on the PC got carried over to the Ally X where I normally cant even configure it.

On PC, I use Hold Guide Button and Gyro as a Quick Mouse setup. Haven't used it in a while, so I forgot about it.
But yeah.

Steam Bullshit strikes again.

Resetting the Guide Button Layout on PC got rid of the Issue on the Ally.
Kinda Stupid but what can you do.

Thanks, anyway, for the time :P

Maybe someone else stumples on this and this saves them alot of time.

closed 2025-02-02 by NightHammer1000