hhd-dev/hhd #141: GPD Win Mini 2024 - HHD has stopped working for controller emulation under gamemode and no controller is detected. Tried latest stable and testing builds of Bazzite Feb 03.

SimpleHeuristics, 2025-02-04:
https://paste.centos.org/view/b6e7bb6d

Can't bring up the HHD overlay in gamemode with either the usual QAM button (R4) or the steam menu with (MENU). Cannot bring up the HHD overlay with Ctrl 3 or Ctrl 4 on keyboard either. Controller is also not detected / emulated. Switching to mouse mode with the switch allows mouse navigation. Switching back to controller no change.

I believe HHD or at least adjuster is still working in the background because my fan curve still seems to be working.

Switching to desktop mode I can launch the HHD UI and seemingly adjust settings but nothing applies to the controller.

Other times when the controller fails to be detected a full shutdown and reboot solves it but it has not in this case.

antheas, 2025-02-04:
Are you in mouse mode? The controller is missing

SimpleHeuristics, 2025-02-04:
No, I toggled between mouse and controller mode. With mouse mode the cursor shows up fine with the right stick and other buttons work too. Once I switch to controller mode it doesn't work anymore.

SimpleHeuristics, 2025-02-05:
Thanks, reflashing the firmware a couple of times in windows and recalibrating the controller did eventually get it to be recognized again.

Seems like a fluke event!

Do you know of any ways to flash firmware / calibrate without windows? I've essentially removed all windows from everything in my life.

closed 2025-02-05 by SimpleHeuristics