hhd-dev/hhd #148: Legion-Go controller always showing even when disconnected.

W0lfwang, 2025-02-17:
When I'm not using the Legion-Go controller I can see the controller listed by steam as "Legion-Controller 1-78". This controller appearing even when the controller is in my house has proven an issue with multiplayer games as some games will recognize it as an input. A few examples:

- In towerfall ascension the game will have problems recognizing one of the controllers, only solved when reordering them on steam. This is not even that straight forward, as it's not always the same position that the "Legion-Controller 1-78" must be moved to, sometime I have to move it to the first input, sometimes as the second and some other times as the last.
- Retroarch needs constant adjustment, but I can understand Retroarch is Retroarch, and it can be trouble by itself.
- Other games it just becomes annoying, if the consoles goes to sleep, I will need to reorder constantly.

I think this is an issue worth checking. Thanks a lot for your attention, and have a nice day.

antheas, 2025-02-17:
When you are not using it? Is half of the controller connected?

W0lfwang, 2025-02-17:
Maybe some further information would be useful. My Legion Go uses Bazzite, the image is the Legion Go & Legion Go S image, with GNOME Desktop. I believe the behavior should be consistent with any Lenovo Legion Go, as I don't believe nothing in my installation could interfere with HDD.

Versions of HDD:
Handheld Daemon Version: 3.12.2
Handheld Daemon UI Version: 3.3.6
Adjustor (TDP) Version: 3.7.7

It's also woth mentioning that this happens even if I have the setting "Controller: Legion Controllers -> Emulation Mode (X-Input): Hidden"

If I can assist with any aditional information, I will be happy to do so.

W0lfwang, 2025-05-04:
Hi, update if anyone finds this issue and want a solution: I wrote a script to solve it, here is the git. It works by unbinding the controller.

https://github.com/W0lfwang/Legion-Go_Disable_controller