hhd-dev/hhd #155: Performance presets causing lag overtime

GUWPPY, 2025-03-12:
I just found out about the per game TDP settings on the QAM but since it sets the TDP using the performance presets when set to a specific number and when I set it to 15w it would set the TDP to balanced mode, over time the game would be very laggy.

It wouldnt lag when using the custom TDP set to 15w though

lilschoolkitten, 2025-03-12:
When I had a legion go I would notice this too. On my Ally X, I have observed it too but recycling tdp limit through SteamOS seems to temporary resolve it

antheas, 2025-06-30:
> I just found out about the per game TDP settings on the QAM but since it sets the TDP using the performance presets when set to a specific number and when I set it to 15w it would set the TDP to [...]

You might be overheating. Using the presets uses a gentler fan curve which makes for a better experience.