hhd-dev/hhd #156: ONEXPLAYER F1 PRO OLED's two shoulder buttons are not functioning.

Decbuan, 2025-03-16:
The ONEXPLAYER F1 PRO OLED's two additional shoulder buttons are not functioning in the latest stable version, and setting them to elite input does not work either. Please request a fix for this issue.

[image]

[image]

antheas, 2025-03-16:
Was it working before? When did this change?

Decbuan, 2025-03-16:
> Was it working before? When did this change?

I just installed Bazzite today, but before that, my older version of HDD was working fine on ChimeraOS. Is there anything I can do to help you fix this issue?

Decbuan, 2025-03-16:
> Was it working before? When did this change?

I previously tested using the command `sudo HHD_DEBUG=1 hhd --user $(whoami)`, and there was no output signal from the two shoulder buttons.

Decbuan, 2025-03-16:
> Was it working before? When did this change?

Please let me know if there is anything I can do to help you fix this issue.

antheas, 2025-03-16:
I have not touched the onexplayer code since this commit https://github.com/hhd-dev/hhd/commit/6a77fd4411a8ddcc04b65d3c506475c2a8f03792 in the beginning of december

Decbuan, 2025-03-16:
> I have not touched the onexplayer code since this commit [6a77fd4](https://github.com/hhd-dev/hhd/commit/6a77fd4411a8ddcc04b65d3c506475c2a8f03792) in the beginning of december

Perhaps I can try testing with an older kernel and the latest HDD to see if it's a kernel-related issue.

antheas, 2025-03-16:
Yes, try something from early in the year. And make sure you are in the last version first off.

Decbuan, 2025-03-18:
> 是的，从年初开始尝试一些东西。首先确保你使用的是最新版本。

I downgraded the kernel to version 6.12.6, and it is functioning properly with the latest hhd 3.13.5 release. Therefore, the issue appears to be caused by subsequent kernel updates. What would be a good solution for me to address this?

antheas, 2025-03-18:
Do some more tests. The top buttons are not controlled by the kernel. So it would be a weird regression, probably with serial handling.

Also, another user on the bazzite discord said they work

antheas, 2025-08-20:
This should have been fixed last week. It was because the f1 pro has two serial ports and the wrong one was used.