hhd-dev/hhd #157: Feature request.  Suspend on overheat

fennectech, 2025-03-19:
I’d like an option to suspend the system at a desired temperature. This would be very helpful for those using cases that seal the vents.  for example id like to suspend my Lenovo legion go at 98C   This would allow overheating to be more gracefully handled without the need for a shutdown

antheas, 2025-03-19:
This is a well meaning suggestion. Unfortunately, the Legion Go overheats at normal temperatures, so I am unsure if that would help.

Also, in custom mode it is programmed to go to 100C so there is a chance you end up there anyway without blocking the fan.

fennectech, 2025-03-19:
that’s why allowing the user to customize the option and / or turn it off is important.    I usually run arround 85-95C  so I don’t think my unit is throttling.

antheas, 2025-06-30:
I have had my legion go overheat at lower temperatures. I'd suggest using Full Fan Speed mode to counteract this now that it is summer.