hhd-dev/hhd #158: Installer fails on official steamos

jorgechavezxyz, 2025-03-25:
### ### 

### [image]

YackMc, 2025-03-27:
I ran into the same issue and it took me forever to actually get it sorted, I'm assuming it should be the same steps (I'm running it on the ROG Ally)

I've no idea where you're staring from, but working with the basis that you already have Decky installed and a user password set:

I followed [these directions ](https://github.com/Jguer/yay/issues/2098#issuecomment-1505042709) to install yay-bin but with an extra step

```
# Allow to write over file system (steam os block)
sudo steamos-readonly disable

# Ensure we have the keys updated
sudo pacman-key --init
sudo pacman-key --populate archlinux

# Install git and base-devel dependencies
sudo pacman -S --needed git base-devel
# Ensure that dependencies get installed correctly (this may be not necessary in some cases)
sudo pacman -S git base-devel
# When prompted, select all options

INSTALL FROM BINARY
git clone https://aur.archlinux.org/yay-bin.git
cd yay-bin
pacman-key --populate holo
pacman-key --populate archlinux
makepkg -si

# Disable again the write over file system
sudo steamos-readonly disable 
```
This allowed me to install yay-bin.

Then I ran the dist install using 

`yay -S hhd adjustor hhd-ui`

Then went to local repo version by running

```
git clone https://github.com/hhd-dev/hhd
cd hhd
python -m venv venv
source venv/bin/activate
sudo pacman -S glibc linux-api-headers
update with pip install --upgrade pip
pip install -e .
```

I was then able to get an output and the ability to connect to hhd through `http://localhost:5335` by running

`hhd`

I skipped the udev rules and modprobe (honestly have no idea what they do)

I then installed hhd decky using the quick install method

`curl -L https://github.com/hhd-dev/hhd-decky/raw/main/install.sh | sh
`

I'm not sure if there's any point in it but I did run 

```
sudo systemctl enable hhd@$(whoami) 
sudo systemctl restart plugin_loader.service
```

Once rebooted, I had hhd working in both desktop and steam gaming mode

YackMc, 2025-03-27:
Just on this too

the error log shows that the pip version is the issue there, if you run `pip install --upgrade pip` it should let you get further

antheas, 2025-06-30:
You should use bazzite if you want to use hhd. Unfortunately it requires a lot of kernel patches to make it realistic to run on steamos.

```
curl -L https://github.com/hhd-dev/hhd-decky/raw/main/install.sh | sh
``` do not use the decky plugin

closed 2025-06-30 by antheas