hhd-dev/hhd #161: Incompatibility with Steam OS 3.8 April 2nd Update

daltonscott23, 2025-04-03:
Steam OS 3.8 users that installed Steam OS on their Legion Go have been having issues with Handheld Daemon running on their devices since an update of Steam OS on April 2nd.

After the update, users get this error when running the hhd command in konsole:

**“Could not start http api on port 5335
Is another version of Handheld Daemon open?”** 

And then from there the API begins to shut down everything.

I have tried both changing the port number as well as reinstalling Steam OS as a whole and then installing Handheld Daemon once more to be greeted with this error message.

Also, to clarify, Handheld Daemon did indeed work before this April 2nd update.

Any help is appreciated! Ty!

![Image](https://github.com/user-attachments/assets/c64e17d1-bead-42e8-

BGuldhammer, 2025-04-04:
Is does not work with latest steam version 3.8 1743720314

auggie246, 2025-05-10:
same issue with hhd on 3.7.5 beta

```
REST   ERROR    Could not start http API on port 5335.                                                        
                Is another version of Handheld Daemon open?                                                   
                Closing.                          
```

antheas, 2025-05-10:
Inputplumber etc. Make sure none of that is running. You're also not going to have a good time with the SteamOS kernel and the ally x