hhd-dev/hhd #167: L4/R4 buttons doesn't work on GPD Win Mini 2025

OuinOuin74, 2025-04-24:
I just received my new Win Mini 2025 (HX370) and everything works as expected exept for L4 and R4 buttons who seems not be mapped.

More info here : https://github.com/aarron-lee/gpd-win-tricks/issues/8#issuecomment-2728278813

alexbowe, 2025-04-28:
Is `winshark` something that comes with GPD's WinControls?

I believe WinControls is not updated for the Win Mini 2025 yet (Not sure what the ETA is - GPD has been saying they will update it for a while now).

Is there anything else that might be able to capture input device signals?

JoeysRetroHandhelds, 2025-05-18:
+1 - L4/R4 don't seem to work on my Win Mini HX370.

HHD unfortunately won't open, but I've remapped it for now.

If I can help provide anything, let me know!

antheas, 2025-05-18:
> Is `winshark` something that comes with GPD's WinControls?
> 
> I believe WinControls is not updated for the Win Mini 2025 yet (Not sure what the ETA is - GPD has been saying they will update it for a while now).
> 
> Is there anything else that might be able to capture input device signals?

If they don't work in Windows, how will they work in Linux?

If one of you is feeling adventurous:

Update those to match your device (`sudo hhd.contrib hidraw`)
https://github.com/hhd-dev/hhd/blob/13e3b8a830ca35e0a794707014fbe5a7abc869b2/src/hhd/device/gpd/win/wincontrols.py#L502

Then, after you see a version error, add the supported version:
https://github.com/hhd-dev/hhd/blob/13e3b8a830ca35e0a794707014fbe5a7abc869b2/src/hhd/device/gpd/win/wincontrols.py#L373-L374

Keep in mind that the controller is very sensitive, and if the protocol is different the chances you will end up bricking it are **not zero**. The only proper way is to capture Wincontrols in Windows.