hhd-dev/hhd #170: HHD daemon fail to launch with error: "ValueError: invalid literal for int() with base 10: '800MHz'"

SamlRx, 2025-05-11:
I am running latest version of Bazzite on a RogAllyX plugged to an eGPU.

Since yesterday, while I am starting my Gnome session HHD is not running.

I have been trying to run it manually using :

```shell
hhd --user MY_USER
```
But HHD exit with this error

```shell
╭─────────────────────────────── Traceback (most recent call last) ────────────────────────────────╮
│ /usr/bin/hhd:8 in <module>                                                                       │
│                                                                                                  │
│   5 from hhd.__main__ import main                                                                │
│   6 if __name__ == "__main__":                                                                   │
│   7 │   sys.argv[0] = re.sub(r"(-script\.pyw|\.exe)?$", "", sys.argv[0])                         │
│ ❱ 8 │   sys.exit(main())                                                                         │
│   9                                                                                              │
│                                                                                                  │
│ /usr/lib/python3.13/site-packages/hhd/__main__.py:333 in main                                    │
│                                                                                                  │
│   330 │   │   │   │   except Exception as e:                                                     │
│   331 │   │   │   │   │   logger.warning(f"Could not hide update settings. Error:\n{e}")         │
│   332 │   │   │   │   settings = merge_settings(                                                 │
│ ❱ 333 │   │   │   │   │   [hhd_settings, *[p.settings() for p in sorted_plugins]]                │
│   334 │   │   │   │   )                                                                          │
│   335 │   │   │   │   # Force general settings to be last                                        │
│   336 │   │   │   │   if "hhd" in settings:                                                      │
│                                                                                                  │
│ /usr/lib/python3.13/site-packages/adjustor/drivers/gpu/__init__.py:100 in settings               │
│                                                                                                  │
│    97 │   │   │   self.core_available = False                                                    │
│    98 │   │   │   return {}                                                                      │
│    99 │   │                                                                                      │
│ ❱ 100 │   │   status = get_igpu_status()                                                         │
│   101 │   │   if not status:                                                                     │
│   102 │   │   │   self.core_available = False                                                    │
│   103 │   │   │   if not self.logged_error:                                                      │
│                                                                                                  │
│ /usr/lib/python3.13/site-packages/adjustor/fuse/gpu.py:87 in get_igpu_status                     │
│                                                                                                  │
│    84 │   │   with open(os.path.join(hwmon, GPU_FREQUENCY_PATH), "r") as f:                      │
│    85 │   │   │   for line in f.readlines():                                                     │
│    86 │   │   │   │   if line.startswith("0:"):                                                  │
│ ❱  87 │   │   │   │   │   freq = int(line.split()[1].replace("Mhz", ""))                         │
│    88 │   │   │   │   if line.startswith("SCLK"):                                                │
│    89 │   │   │   │   │   freq_min = int(line.split()[1].replace("Mhz", ""))                     │
│    90 │   │   │   │   │   freq_max = int(line.split()[2].replace("Mhz", ""))                     │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
ValueError: invalid literal for int() with base 10: '800MHz'
```

Is there anywhere I can look?

antheas, 2025-05-11:
Indeed. Thanks. Although your bug is double. That code should be finding your iGPU only, but it seems to be finding your dGPU. If you can find why the check in `find_amd_igpu()` fails that would be great.

https://github.com/hhd-dev/adjustor/releases/tag/v3.10.4

antheas, 2025-05-11:
> I tried to submit a pull request, but it seems I don't have the necessary permissions.

You need to fork the repository, then push your branch there, and then create a pull request. You cannot push branches to the repository itself directly. If you did that I can look into it, although I have not changed any permissions.

closed 2025-05-11 by antheas

SamlRx, 2025-05-11:
Done here: https://github.com/hhd-dev/adjustor/pull/15