hhd-dev/hhd #171: GPD WIN4 2025 gyro axis issue

johnisyonman, 2025-05-17:
I'm using the GPD Win 4 (2025 HX370), and I've noticed that the gyro is initially set to portrait mode by default.
To correct this, I applied the following axis override:

X → Z (Inverted)
Y → X
Z → Y (Inverted)

At first, the override seemed to work well, but after a few minutes, the gyro starts to drift diagonally. For example, when I tilt the device upward along the X-axis, the movement registers as up and to the left—and the same happens in reverse when tilting downward.
This diagonal shift doesn’t occur in the default mode, but the axes are completely wrong there, so it’s not usable either.

Is there a way to fix this issue? Any advice would be greatly appreciated!

johnisyonman, 2025-07-03:
[image]

G1618-04 it is.
Let me know if you need anything else

imtpot, 2025-10-01:
The same problem with GPD win mini 2025 (G1617-02)