hhd-dev/hhd #179: Option to disable controller handling

Faupi, 2025-06-07:
Motivation:
When HHD is used alongside InputPlumber, HHD just fails every 3 seconds at initializing the controller, causing various input hang-ups in e.g. Moonlight.

As handheld-daemon has quite a few options and often those are able to be disabled (TDP, RGB, etc.) and with SteamOS (or maybe just gamescope) requiring InputPlumber lately, I think it would make sense to have an option to also outright disable controller handling on hhd.

Coming from NixOS, with a SteamOS-like setup via [Jovian-NixOS](https://github.com/Jovian-Experiments/Jovian-NixOS), using InputPlumber seems to be quite stable and doesn't require additional setup, as via handheld-daemon - however I have not found a good alternative for other features, so it would suck to have to switch just because of this.

Maybe the simplest solution would be to be able to disable specific plugins, which would be more general - in my case I would just choose to disable `legion_go_controllers`

For reference, these are the logs that get spammed every 3s, due to the device grabbing:
```
Jun 07 21:55:35 go hhd[942]: LEDS   INFO     Setting RGB to mode 'solid' with color: 127,  63,   0.
Jun 07 21:55:36 go hhd[942]: LLGO   INFO     Launching emulated controller.
Jun 07 21:55:36 go hhd[942]: LLGO   ERROR    Received the following error:
Jun 07 21:55:36 go hhd[942]:                 <class 'OSError'>: [Errno 16] Device or resource busy
Jun 07 21:55:36 go hhd[942]: LLGO   ERROR    Assuming controllers disconnected, restarting after 3s.
```
Startup logs (hhd-ui coredump omitted due to faulty setup of that component):
```
Jun 07 21:54:50 go systemd[1]: Started Handheld Daemon.
Jun 07 21:54:50 go hhd[942]:    _______   _______    ______
Jun 07 21:54:50 go hhd[942]:   /    /  ╲╲/    /  ╲╲_/      ╲╲
Jun 07 21:54:50 go hhd[942]:  /        //        //        //
Jun 07 21:54:50 go hhd[942]: /         /         /         /
Jun 07 21:54:50 go hhd[942]: ╲___/____/╲___/____/╲________/
Jun 07 21:54:50 go hhd[942]: MAIN   INFO     Handheld Daemon starting...
Jun 07 21:54:50 go hhd[942]: MAIN   INFO     Running under an unknown Linux distro.
Jun 07 21:54:50 go hhd[942]: MAIN   INFO     Running autodetection...
Jun 07 21:54:51 go hhd[942]: MAIN   INFO     Found plugin providers: bootc, debug, generic, gpd_win,
Jun 07 21:54:51 go hhd[942]:                 legion_go, msi_claw, onexplayer, orange_pi, overlay, power,
Jun 07 21:54:51 go hhd[942]:                 powerbuttond, rgb, rog_ally, adjustor
Jun 07 21:54:51 go hhd[942]: MAIN   INFO     Lenovo BIOS version: 37
Jun 07 21:54:51 go hhd[942]: MAIN   INFO     Loaded the following plugins:
Jun 07 21:54:51 go hhd[942]:                   - legion_go: legion_go_controllers
Jun 07 21:54:51 go hhd[942]:                   -  overlay: overlay
Jun 07 21:54:51 go hhd[942]:                   -    power: power
Jun 07 21:54:51 go hhd[942]:                   - powerbuttond: powerbuttond@'Legion Go'
Jun 07 21:54:51 go hhd[942]:                   -      rgb: controller_rgb
Jun 07 21:54:51 go hhd[942]:                   - adjustor: adjustor_lenovo, adjustor_init, adjustor_main,
Jun 07 21:54:51 go hhd[942]:                 adjustor_battery, adjustor_ppd
Jun 07 21:54:51 go hhd[942]: MAIN   INFO     Loaded the following locales:
Jun 07 21:54:51 go hhd[942]:                  - adjustor (20):
Jun 07 21:54:51 go hhd[942]:                 /nix/store/zizrag13d4ic8cr2q8d21islzbvzlkr2-handheld-daemon-3.15
Jun 07 21:54:51 go hhd[942]:                 .3/lib/python3.12/site-packages/hhd/i18n/./
Jun 07 21:54:51 go hhd[942]:                  - hhd (10):
Jun 07 21:54:51 go hhd[942]:                 /nix/store/zizrag13d4ic8cr2q8d21islzbvzlkr2-handheld-daemon-3.15
Jun 07 21:54:51 go hhd[942]:                 .3/lib/python3.12/site-packages/hhd/i18n/./
Jun 07 21:54:51 go hhd[942]: POWR   INFO     Found battery:
Jun 07 21:54:51 go hhd[942]:                 /sys/class/power_supply/BATT
Jun 07 21:54:51 go hhd[942]: MAIN   INFO     Reloading configuration.
Jun 07 21:54:51 go hhd[942]: MAIN   INFO     Handheld Daemon Version: 3.15.3
Jun 07 21:54:51 go hhd[942]: MAIN   INFO     Adjustor Version: 3.10.3
<hhd-ui coredump>
Jun 07 21:54:52 go hhd[942]: MAIN   INFO     Overlay Version: 3.3.6
Jun 07 21:54:52 go hhd[942]: MAIN   INFO     No profiles found.
Jun 07 21:54:52 go hhd[942]: MAIN   INFO     Initialization Complete!
```

antheas, 2025-06-07:
Inputplumber is not required for gamescope?

What's the usecase for having it?

antheas, 2025-06-07:
So if you disable controller support in hhd you will lose all controller related functionality. So there is no point in using it.

You should contact the jovian maintainers to add an option to switch

Steamos manager will also conflict with tdp settings for now, but we will patch it soon

Faupi, 2025-06-07:
I personally use HHD for power profile switching (ideally for desktop use with the existing powerdevil emulation too, just have to fix my configuration), fan curves, power light, and TDP management as a whole, so I think even just besides the controller there are still plenty of usecases.

I'll check in with the Jovian maintainers and see if the InputPlumber requirement could be done more "optionally", but I would still argue there could be some benefit in letting users of HHD opt-out of the controller handling. But maybe it's true that there are a lot of parts of HHD that depend on the controller handling - this I do not know, as I don't really use the overlays or specific shortcuts, as I manage what I have to via the decky plugin and HHD-UI on the desktop currently.

Faupi, 2025-06-08:
I found the blacklist config for the plugins, however it still loads the `legion_go` provider along with controllers:
```
Jun 08 02:31:54 go systemd[1]: Started Handheld Daemon.
Jun 08 02:31:54 go hhd[33207]:    _______   _______    ______
Jun 08 02:31:54 go hhd[33207]:   /    /  ╲╲/    /  ╲╲_/      ╲╲
Jun 08 02:31:54 go hhd[33207]:  /        //        //        //
Jun 08 02:31:54 go hhd[33207]: /         /         /         /
Jun 08 02:31:54 go hhd[33207]: ╲___/____/╲___/____/╲________/
Jun 08 02:31:54 go hhd[33207]: MAIN   INFO     Handheld Daemon starting...
Jun 08 02:31:54 go hhd[33207]: MAIN   INFO     Running under an unknown Linux distro.
Jun 08 02:31:54 go hhd[33207]: MAIN   INFO     Running autodetection...
Jun 08 02:31:54 go hhd[33207]: MAIN   INFO     Skipping blacklisted provider 'legion_go'.
Jun 08 02:31:54 go hhd[33207]: MAIN   INFO     Found plugin providers: bootc, debug, generic, gpd_win,
Jun 08 02:31:54 go hhd[33207]:                 legion_go, msi_claw, onexplayer, orange_pi, overlay, power,
Jun 08 02:31:54 go hhd[33207]:                 powerbuttond, rgb, rog_ally, adjustor
Jun 08 02:31:54 go hhd[33207]: MAIN   INFO     Lenovo BIOS version: 37
Jun 08 02:31:54 go hhd[33207]: MAIN   INFO     Loaded the following plugins:
Jun 08 02:31:54 go hhd[33207]:                   - legion_go: legion_go_controllers
Jun 08 02:31:54 go hhd[33207]:                   -  overlay: overlay
Jun 08 02:31:54 go hhd[33207]:                   -    power: power
Jun 08 02:31:54 go hhd[33207]:                   - powerbuttond: powerbuttond@'Legion Go'
Jun 08 02:31:54 go hhd[33207]:                   -      rgb: controller_rgb
Jun 08 02:31:54 go hhd[33207]:                   - adjustor: adjustor_lenovo, adjustor_init, adjustor_main,
Jun 08 02:31:54 go hhd[33207]:                 adjustor_battery, adjustor_ppd
```

In `plugins.yml` I simply have:
```
blacklist:
- legion_go
```

Is it always forced when it detects the device model/capabilities?

antheas, 2025-06-20:
Depends on the device. On devices that are known to work such as the Legion ones it is forced on. On other devices e.g., Ayaneo it can be disabled. This way, we can autoload the driver even without an exact match