hhd-dev/hhd #195: [Legion Go] Standard TDP profiles (8W, 15W, 30W) completely ignore custom fan curve

azenyr, 2025-07-21:
This would not be a problem if HHD didnt automatically set these standard profiles when setting 8W or 15W via steam QAM. If we manually set 15W via custom profile in HHD, this problem does not happen.

The main problem however is that these standard profiles completely ignore custom/manual fan curves, even though the "custom fan curve" option is still set to manual. Switching the TDP to like 14 or 16W on steam QAM, switches HHD back to custom TDP mode and the fan curve instantly applies (usually fans go from screaming very loud (by default lenovo has a VERY aggressive fan profile) to instantly much quieter (my custom fan profile). If I switch back to exactly 15W (HHD switches to standard 15W balanced mode) the fan RPM starts climbing right away completely ignoring my custom fan curve once again.

Its been like this for at least 2 months now. HHD is updated to the latest stable version via bazzite stable updates (updated via HHD itself).

azenyr, 2025-07-21:
I bought this device used, never even turned windows on on it. Instantly installed bazzite when I took it out of the box. This was like 2 months ago when I bought it, and it already had this bug.
I don't remember which BIOS version it came with, but it was not the latest, and already had this bug. I have since updated via fwupd to version N3CN38WW - the most recent version, released in 03 june - and this bug persists. So the previous BIOS version also had this bug, but I don't remember which version it was.

antheas, 2025-07-21:
You were never supposed to be able to set a fan curve in other modes, so that means that later bioses fixed that and I will need to remove that ability

azenyr, 2025-07-21:
If that's the case, I suggest that we change the "custom fan mode" setting behaviour: when set to "manual", it should always stay at the custom TDP mode, and never go to the standard modes even if we set 8W or 15W on steam QAM. Should only go to standard modes if fan curve is set to default/disabled. That should at least avoid this problem.