hhd-dev/hhd #198: SInput HID Protocol Support

AL2009man, 2025-07-25:
Not to long ago: [Hand Held Legends](https://handheldlegend.com/), the folks behind the upcoming GC Ultimate, introduce a new H[ID Protocol specifically designed for SDL-based Apps and Steam Input](https://github.com/HandHeldLegend/SInput-HID). Addressing the major problem that third-party gamepad faced, allowing Analog Joystick and Motion Sensors at the same time while being a bespoke Controller Type.

Since then: it's already merged onto mainline SDL and SteamInput, it might be a good idea to introduce a new "Controller Emulation" that could become the new default going forward.

antheas, 2025-07-25:
If it supports RGB, dual, quad buttons, and motion, where all are optionally turned on we can do that. It would be great, yes

AL2009man, 2025-07-25:
> If it supports RGB, dual, quad buttons, and motion, where all are optionally turned on we can do that. It would be great, yes

you can view the documents if you'd like to learn more: https://docs.handheldlegend.com/s/sinput/doc/sinput-hid-protocol-dev-SNSaEw36nc

> Also, it needs to support specifying the controller is a handheld. This is important for setting the roll axis correctly for gyro fps aiming

but I'll ask Handheld Legends folks about the Motion Sensor part

AL2009man, 2025-07-25:
> Also, it needs to support specifying the controller is a handheld. This is important for setting the roll axis correctly for gyro fps aiming

so, i asked SInput devs about this part. SInput should always expect a gyro order, so i believe HHD can handle the device-based IMU correction, and then pass it through a virtual SInput device. I'll suggest reading thru the documents for more details,

AL2009man, 2025-07-25:
> We can simplify where appropriate for vid and pid pairs
> 
> E.g. handheld without gyro with 0, 2, 4 extra buttons. And then with gyro. 6 pairs total.

AL2009man, 2025-07-25:
ers

antheas, 2025-07-26:
> so, i asked SInput devs about this part.
>
> SInput should always expect a gyro order, so i believe HHD can handle the device-based IMU correction, and then pass it through a virtual SInput device. [...]

It's difficult to understand if you never have done gyro aiming or had a handheld, but on handhelds the roll direction needs to be reversed compared to a controller when fps aiming. When doing a trick such as moving the Zelda table puzzle it needs to be normal.

So consumers need to know their controller is a handheld screen to account for that

AL2009man, 2025-07-27:
gonna temporarily close it for a bit