hhd-dev/hhd #228: Syntax error on F-string on Python 3.10

songyw124, 2025-09-24:
Good afternoon,

When I am trying to use HHD on Python 3.10, I keep gets problem on F-Strings.
For example, `f"{SD_SETTINGS[rtype] if rtype < len(SD_SETTINGS) else "UKNOWN"} ({rtype:02d}): {rdata:02x}"` on controller/virtual/sd/__init__.py gives syntax error because of nested double quotations.

(By the way, I guess UKNOWN over the codes are a typo of UNKNOWN?)

When I manually change all these occurences in the code, then HHD starts to work.
I believe that HHD supports Python 3.10 according to the documentation, but maybe is this a problem of Python version?

Thank you very much.
Ευχαριστώ πολύ.

antheas, 2025-09-24:
Thank you for finding these occurrences. If you can PR the changes it would be great.

songyw124, 2025-09-24:
Thank you very much for your response. I would like to ask you if my understanding is correct, that the package will have no problems on Python 3.12 – and in that case maybe changing the supporting version from 3.10+ to 3.12+ will be a solution.

Unfortunately, I am worrying about messing up the repository because I am not familiar with GitHub, and will try to do that once I get familiar to it. Or if there are any contributors that can help me regarding this issue, I would be thankful for that.